litex-hub/litespi #5: Add travis CI

mithro, 2020-03-30:
- Should run the tests
 - Should check the style (might be problematic?)

piotr-binkowski, 2020-03-30:
I think that travis is already used for running the tests https://travis-ci.com/github/litex-hub/litespi/builds/156593973
```
test_spi_xip_core_syntax (test.test_spi_xip.TestSPIXIP) ... ok

test_spi_xip_read_test (test.test_spi_xip.TestSPIXIP) ... ok

----------------------------------------------------------------------

Ran 2 tests in 2.424s

OK

The command "python setup.py test" exited with 0.
```
Unless you meant some other tests by that?

mithro, 2020-03-30:
I didn't actually think to look to see if travis was already running, so feel free to close if it already is.

closed 2020-04-07 by enjoy-digital