litex-hub/litespi #23: Support using the SPI controller in a separate clock domain

piotr-binkowski, 2020-04-09:
With that we could run the rest of the SoC at a slower frequency while running the SPI core at a higher frequency to get better SPI/XIP performance. I think that this could be achieved simply by adding AsyncFIFO between SPI cores and the PHY. 
What do you think? @enjoy-digital @mithro

mithro, 2020-04-09:
A AsyncFIFO might cause a large latency hit?

I think you might be able to take advantage of the fact that clocks generated from the same source and phase aligned to reduce CDC delays.

I'm no CDC expert.

kowalewskijan, 2020-04-27:
@enjoy-digital What do you think about this? Is there anything similar implemented in LiteX already?

enjoy-digital, 2020-04-27:
@enjoy-digital: i would recommend using an AsyncFIFO for the CDC, this is the easiest solution and the more generic. If latency is too high and if we have more information about the clocks relationship, we could replace the AsyncFIFO with a module that will take advantage of this and would just optionally replace the AsyncFIFO, so the architecture will be the same.

You can find some similar code here:
- https://github.com/enjoy-digital/liteeth/blob/master/liteeth/mac/core.py#L103-L107
- https://github.com/enjoy-digital/usb3_pipe/blob/master/usb3_pipe/serdes.py#L259-L262
- https://github.com/enjoy-digital/usb3_pipe/blob/master/usb3_pipe/serdes.py#L305-L308

kowalewskijan, 2020-05-28:
@enjoy-digital I understand that we can CDC streams and I think we should separate clkgen too, but how we can handle cross domain rw to CSRs? I mean e.g. these [registers](https://github.com/litex-hub/litespi/blob/master/litespi/__init__.py#L51-L52).

enjoy-digital, 2020-05-29:
@kowalewskijan: sorry i saw your questions but haven't answered directly and then forgot... The  control signals you want to transfer to the PHY's clock domain are slow (configuration from the software) and not moving once configured, so that's probably better to just use MultiReg as done here for example: https://github.com/enjoy-digital/litejesd204b/blob/master/litejesd204b/core.py#L327-L369. If these CSRs are only affecting the PHY, you should also probably create them directly in the PHY.

PR #38 (2020-06-03): Add optional litespi wrapping into separate clock domain
This PR introduces changes which give a possibility to have LiteSPI in a separate clock domain.
The only doubt I have is do we have to pipeline the data from CSRs (which are in sys clock domain) to litespi clock domain using e.g. AsynFIFOs? I mean these [registers](https://github.com/litex-hub/litespi/pull/38/files#diff-76745ba1ce398bffc095fd8450e49fcdL51-L52). I tested on Arty board two configurations:
- LiteSPI in `sys` clock domain
- LiteSPI in `clk200` clock domain

in each cases I was able to successfully read from flash.

Closes #2 [... cut]
closed 2020-06-03 by enjoy-digital