litex-hub/litespi #25: Support the SPI RAM found on HAD badge

mithro, 2020-04-10:
The [HackADay badge](https://hackaday.com/2019/11/04/gigantic-fpga-in-a-game-boy-form-factor-2019-supercon-badge-is-a-hardware-siren-song/) has
> two Lyontek LY68L6400 64 Mbit SPI SRAM chips

These SPI SRAM chips, *kinda* look like SPI NOR chips.

It would be good to have these devices supported as memory mappable for both read and write.

Clock rate:
- 33MHz(max) for normal read
- 100MHz(max) for fast read
- Mode: SPI/QPI

Datasheet -> https://www.electrodragon.com/w/images/0/04/LY68L6400_0.3.pdf

machdyne, 2022-07-24:
Hi @mithro. I have been trying to get memory-mapped (Q)SPI PSRAM working with LiteX for some time now with no success. I have some working Verilog for QSPI PSRAM in this repo: https://github.com/machdyne/qqspi

If anyone wants to attempt this I can probably provide a 32MB QSPI PSRAM PMOD.

xobs, 2022-07-24:
I had this working in https://github.com/xobs/haddecks/blob/master/rtl/spi_ram.py, and I was working on interleaving access in https://github.com/xobs/haddecks/blob/master/rtl/spi_ram_dual.py

There are tests in https://github.com/xobs/haddecks/blob/master/sim/spiram.v

machdyne, 2022-07-25:
Thanks! This is working for me. I had to use dummy=6 with APS6404L-3SQR-SN PSRAM.

Maybe this can be merged into litex.soc.cores or litespi?

I have noticed a few boards with similar SPI PSRAM and with the QQSPI PMOD it's possible to add 32MB to any board with a PMOD socket, having this capability in LiteX might be useful, especially for the smaller boards.

suarezvictor, 2023-02-27:
I wonder why this is not merged if working. Do you think it wouldn't be hard to add it to the reference board script? Hackaday_badge.py
It currently expects an external SDRAM chip.
Is there any migen example for machdyne board/PMODs?

suarezvictor, 2023-02-28:
This development makes the PSRAM appear as SPI flash https://github.com/litex-hub/litex-boards/pull/476/commits/b1df2c0f85fbcd5d4880ae71c05ac39b0abc4291

PR #70 (2024-07-31): mmap: Add write functionality
With this change, it is possible to memory map e.g. PSRAM. When setting with_write to "csr", it can also be used to support faster flash writing via etherbone. For flash, only False and "csr" should be used, as the some other logic or application needs to make sure that the flash is erased ahead of time. True is only meant for use with RAM chips.

There are two bugs that this also addresses:
1. ~~Addresses are now masked to flash size or addr_width if no size is set. I'm not 100% happy with this solution, as it might cause issues with non power of 2 flashes (do th [... cut]
closed 2024-07-31 by enjoy-digital