litex-hub/litespi #39: Fix the name of LiteSPI instance in example/arty.py

kowalewskijan, 2020-05-29:
I see that in `examples/arty.py` we use `spiflash_mmap` name for LiteSPI instance ([link](https://github.com/litex-hub/litespi/blob/master/examples/arty.py#L90-L93)). In my opinion it could be quite confusing since LiteSPI by default is configured to have both Master and MMAP mode ([link](https://github.com/litex-hub/litespi/blob/master/litespi/__init__.py#L44)). I think we should decide either we change the name to something more generic like `spiflash_core` or we should remove default master mode.

enjoy-digital, 2020-05-29:
In the long term, `spiflash ` would be probably be the best but we just have to be sure it we will not have software conflicts (in the definitions) with the old SPIFlash core.

enjoy-digital, 2021-07-29:
LiteSPI integration is now directly included in LiteX and the name has been changed to `spiflash_core` with https://github.com/enjoy-digital/litex/commit/c5b5abd1032d26879f88f0a584568935836d0937. For consistency with the others cores we should just name it `spiflash` but this is not possible now due to the conflict is will create with the previous SPI Flash core. This will be done later.

closed 2021-07-29 by enjoy-digital