litex-hub/litespi #40: Support for ECP5 boot flash in LiteSPI

derekmulcahy, 2020-07-23:
Currently clkgen.py has support for xc7 devices using a Xilinx STARTUPE2 instance. Similar support could be added for ECP5 using the Lattice USRMCLK instance. From looking at other code, without understanding much about what is required, it looks like USRMCLK does a similar thing to STARTUPE2. I added the following in clkgen.py and it seems to work.

```
            elif device == "ecp5":
                self.specials += Instance("USRMCLK",
                    i_USRMCLKI  = clk,
                    i_USRMCLKTS = 0
                )

```

This code is selected by passing the`device = "ecp5"` parameter to `LiteSPIPHY`.

I can try the whole pull request process if required.

mithro, 2020-07-23:
@derekmulcahy - Would love to see a pull request, that way you get credit in the git history!

enjoy-digital, 2020-07-23:
@derekmulcahy: indeed that's similar to the `STARTUPE2` primitive, that's what we are doing in the old SPIFlash core: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/spi_flash.py#L56-L78. That would be nice to have a PR for this yes, for the selection, could you just make sure it will use the device name from the Platform (check for `LFE5U`).

derekmulcahy, 2020-07-23:
The updated change is
```
            elif device[:5] == "LFE5U":
                self.specials += Instance("USRMCLK",
                    i_USRMCLKI  = clk,
                    i_USRMCLKTS = 0
                )

```
and the parameter to LiteSPIPHY is now `device  = platform.device`. I needed to drive it in Master mode to update the flash as well as MMAP mode to read it. Both modes are now working well.

I will pull together a PR and include an example for the Colorlite 5A-75E.

PR #41 (2020-08-11): ECP5 support and Colorlight example.
This PR includes support for the USRMCLK instance in clkgen.py and provides an example for the Colorlight 5A-75E. It borrows heavily from [examples/arty.py](https://github.com/litex-hub/litespi/blob/master/examples/arty.py) and [enjoy-digital/colorlite.py](https://github.com/enjoy-digital/colorlite/blob/master/colorlite.py).

Fixes #40
closed 2020-08-11 by piotr-binkowski