litex-hub/litespi #49: Arty: Example for storing gateware on SPIflash

dayjaby, 2021-06-03:
Hi,

has anyone accomplished to store the gateware on the SPIflash on an Arty A7 already? Do we have a documentation available for this? What is the most reasonable way to make it very easy to use (e.g. `--store` instead of ``--load`` after building the gateware?)

Looking forward to your ideas!

dayjaby, 2021-06-03:
Oh, there is ``--flash`` already: https://github.com/enjoy-digital/litex/pull/305

I will try it later on the Arty board and report my results here :) Leaving it open till then

enjoy-digital, 2021-06-03:
Hi @dayjaby,

yes, this is maybe not enabled currently on the Arty but you can look at the Acorn:
- https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sqrl_acorn.py#L170
- https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sqrl_acorn.py#L202-L204

enjoy-digital, 2021-07-29:
Closing since answered and more related to LiteX-Boards.

closed 2021-07-29 by enjoy-digital