litex-hub/litespi #54: Improve resources usage.

enjoy-digital, 2021-07-30:
LiteSPI is working correctly with good performance on the different boards it has been tested but uses more resources than the LiteX SPIFlash core that was used previously.

When testing different configurations on the iCEBreaker (with `python3 -m litex_boards.targets.1bitsquared_icebreaker --cpu-type=serv --build`) we get the following resource usage:

| No SPIFlash | LiteX's SPIFlash | LiteSPI (no Master) | LiteSPI (with Master (depth=1)))
|--|--|--|--|
| 1883 LCs  | 2036 LCs | 2353 LCs | 2921 LCs
| Ref  | +153 LCs | +470 LCs | +1038 LCs
| -  | Ref | +317 LCs | + 885 LCs

The LiteX's SPIFlash core was equivalent to LiteSPI with Master enabled, so we should reduce resource usage to be able to do efficient designs on small FPGAs.

zyp, 2021-07-30:
I've been thinking that the PHY can be simplified a lot by removing the `CMD` and `READ` special casing and having `LiteSPIMMAP` use the `USER` interface instead. There doesn't seem to be any performance benefit in the special casing. While some of the resource usage would just move from the PHY core to the MMAP core, I believe the net total should be lower.

enjoy-digital, 2021-07-30:
@zyp: This architecture change could indeed make senses, it would probably reduce complexity and then resource usage.

Without modifying the architecture, I did some simplifications in the code and the current resource usage status now is:

| No SPIFlash | LiteX's SPIFlash | LiteSPI (no Master) | LiteSPI (with Master (depth=1)))
|--|--|--|--|
| 1883 LCs  | 2036 LCs | 2318 LCs | 2892 LCs
| Ref  | +153 LCs | +435 LCs | +1009 LCs
| -  | Ref | +282 LCs | +856  LCs

zyp, 2021-07-30:
Out of curiosity, have you checked the resource usage with only the master module enabled, not MMAP?

enjoy-digital, 2021-07-30:
With `self.add_spi_flash(mode="4x", module=W25Q128JV(Codes.READ_1_1_4), with_mmap=False, with_master=True)` (and minor tweaks in the code to allow the design to compile) I get `2673 LCs`.

mithro, 2021-07-30:
@kgugala - Can you please get someone to take a look at this?

tmichalak, 2021-08-03:
@pawelsag please take a look