litex-hub/litespi #66: Support Auto-Initializing SPI Flashes to Quad Mode

cr1901, 2021-12-17:
This applies to my version of the [icebreaker](https://github.com/icebreaker-fpga/), and is a follow up of a [conversation](https://libera.irclog.whitequark.org/litex/2021-10-20#1634725337-1634726101;) with @smunaut.

Some SPI flashes, like _some_ part numbers in the W25Q128JV family, support `4x` operation, but are not enabled for `4x` on power-on. My icebreaker uses one of these flashes where the default is `1x` on power on, and therefore `4x` operation does not work out-of-the-box. This can be surprising behavior, and I think initializing to `4x` mode if a user requested a `4x` may be a user-invisible solution :D!

The PHY [supports](https://github.com/litex-hub/litespi/blob/master/litespi/phy/generic_sdr.py#L128-L137) doing `1x` reads even if you supply pads for `4x` operation. I think the easiest way to add support for this is to add some init steps to the `LiteSPIMMAP` FSM, which does the following (based on W25Q128JV datasheet):

1. Send command `50h` (Write Enable for Volatile Status Register)
2. Read SR2 (command `35h`)
3. Set `QE` bit (bit 1) in the read value
4. Write SR2 (command `31h`)
5. Wait 50 nanoseconds (tSHSL2- part-specific?) before allowing _any_ further accesses to take place.

After these 5 steps have occurred, your SPI flash will be in quad mode until power is cycled.

However, I haven't done this myself because I'm not sure the best way to represent "flash _may_ need to be initialized into `4x` mode" per `SpiNorFlashModule`. In the case of W25Q128JV, different part numbers have different defaults (although initializing a part that doesn't need to be initialized is harmless). There's plenty of ways to represent "needs 4x init", but which way is best? Can it be done without regenerating the entirity of `generated_modules.py`?

zyp, 2021-12-17:
The location of the QE bit tends to vary from vendor to vendor, so your sequence is Winbond specific.

Considering the QE bit is also non-volatile writable, I would argue that it generally makes more sense to just write it once (e.g. during factory testing/initialization/first flash write) so that it'll always power on in the right state.

enjoy-digital, 2021-12-17:
@cr1901; As @zyp, I also consider setting QE bit is part of the factory configuration. I'm also not sure how specific to each SPI Flash this configuration sequence can be.  On most  FPGA boards, this bit also already has to be configured to be able to load the bistream in QSPI Mode, so  I'm not sure it's really worth adding this complexity to `LiteSPI`. Wasn't it possible to just configure QE bit permanently on your W25Q128JV SPI Flash?

cr1901, 2021-12-17:
>Wasn't it possible to just configure QE bit permanently on your W25Q128JV SPI Flash?

Yes, and I didn't read the datasheet closely enough (emphasis mine):

> When the QE bit is set to
a 1 (factory fixed default for part numbers with ordering options “IQ/IN” & “JQ”), the Quad IO2 and IO3
pins are enabled, and /HOLD function is disabled, **the device operates in Standard/Dual/Quad SPI modes.**

Meaning it doesn't matter if the FPGA reads the bitstream in `1x` mode if `4x` mode is enabled- everything should still (and does, in my case) work. I set the bit permanently as the solution.

closed 2021-12-17 by cr1901