litex-hub/litespi #78: clkgen from STARTUPE3 and more

jersey99, 2024-10-10:
Hi @enjoy-digital,

Do you have any plans to add STARTUPE3 here? Let's say, yes, then do you have an idea to handle the 4xSPI bits coming from the STARTUPE3 primitive to access the primary SPI flash?

https://github.com/litex-hub/litespi/blame/4f60633cbe744701bf552c5d5a1c404a71581e73/litespi/clkgen.py#L112

I see that you did something similar in the 1x mode inside USSPIFlash, in spi_flash.py in the Litex repo.

I do have a use case for accessing the primary SPI flash potentially reprogramming it via TFTP/Liteeth. I think it will be rather slow over 1x. So I am looking for options here, and have a design in mind. Wondering if you have done this already and/or have something in mind.

Cheers!

jersey99, 2024-10-10:
To be more specific, I can see expanding clkgen to include STARTUPE3 and passing the SPI "pads" through it and assigning the dq_{io/o/i} signals to it etc. I didn't want to get started on re-jigging the whole thing in case that isn't the way you see fit.

enjoy-digital, 2024-10-10:
Hi @jersey99;

interface speed is generally not the bottleneck when writing to SPI Flash since you generally spend a lot more time to waiting for the SPI Flash to be ready after the write, ex https://github.com/enjoy-digital/litepcie/blob/master/litepcie/software/user/liblitepcie/litepcie_flash.c#L238-L240.
 
So I would first recommend starting with x1 which should already be fast enough if you try to group the access in bursts.

jersey99, 2024-10-10:
Sounds good and make sense. Thanks for that.

Does it still make sense to modify the STARTUPE3 instance inside clkgen.py in order to get access to the MMAP and all the features for litespi? The spi_flash inside Litex doesn't seem to have this nice MMAP feature which I see being used from the software side at least.

I do have another (secondary) flash chip on the board (HTG940) currently working with 4x where I don't route the DQ pins via STARTUPE3.

jersey99, 2024-10-11:
@enjoy-digital 
Looks like with this quick hack: https://github.com/jersey99/litespi/blob/test-spi-write/litespi/clkgen.py
The 4x Reads from Flash are working.

Let me know if you are interested in doing something like this. I can clean this up and send a PR.

enjoy-digital, 2026-10-21:
Closing since #108 (STARTUPE3 backend / LiteSPIXilinxUSPHY) has been merged, thanks! Feel free to reopen if something is missing.

closed 2026-10-21 by enjoy-digital